roema/Open-Glider-Network-Groundstation #1: Receive no  traffic

ubx, 2020-11-07:
Hi

Installed version 0.0.7 following the Easy Installation instruction. I have a FLARM (version 7.01) and a OGNTP device nearby:

- No traffic received
- Receive periodic UDP debugging messages (in my case over address  192.168.1.200)

[image]

BTW very interesting project!

roema, 2020-11-07:
Hi Andreas!You need GPS fix.In picture, i see gnss zero.Wait for gps fix, gnss should show you the number of satellites.Then, decoding packets is possible and the groundstation is registering in ogn network. At this point, you should See aprs messages in port 12000.Please let me know, if it works vor not!Thanks! Manu

roema, 2020-11-07:
Hi Andreas!

If i take a look at glider radar, i can see your station with last forwarded aircraft. Is your problem solved?

ubx, 2020-11-07:
Hi Manu

I moved the ground station (Walpwil-M) outside and now it sees 7 satellites. I also turned off my other Raspi ground station (Walpwil), to avoid confusion. But still not traffic forwarded to the ogn network.

[image]

roema, 2020-11-07:
Hi Andreas!

Please test to change "Ignore track bit" and "Ignore stealth bit" to true.

Manu

ubx, 2020-11-07:
Done, still not traffic forwarded.

roema, 2020-11-07:
Your ogn or flarm transponder has gps?

roema, 2020-11-07:
Ok, no i can see you traffic. But there is a bug in ARPS coding.

Error, This is not a valid APRS packet: FLRC424E4>APRS,qAS,Walpwil-M:/104147h4703.70N00713.70En000/000/A=001555 !W94! id3EC424E4 +000fpm +0.0rot 82.00dB 0e -0.0kH

I will fix this today evening.

ubx, 2020-11-07:
Thanks Manu

FYI on UDP debugging I can now see these messages:

[image]
[image]

roema, 2020-11-07:
I have found a BUG in APRS symbol table.
Can you test the attached firmware?
[ognbase.ino.esp32_test.zip](https://github.com/roema/Open-Glider-Network-Groundstation/files/5504685/ognbase.ino.esp32_test.zip)

Thanks, Manu

ubx, 2020-11-07:
Hi Manu

Updated to the new firmware. No periodic UDP debugging messages anymore. Timestamp field doesn't change.

[image]

roema, 2020-11-07:
Have found a bug in aprs symbol table.
Can you test this image?

[ognbase.ino.esp32.zip](https://github.com/roema/Open-Glider-Network-Groundstation/files/5504758/ognbase.ino.esp32.zip)

 Manu

roema, 2020-11-07:
Hi Andreas!

I habe uploaded wrong image, sorry. Please use this: 
[ognbase.ino.esp32.zip](https://github.com/roema/Open-Glider-Network-Groundstation/files/5504762/ognbase.ino.esp32.zip)

ubx, 2020-11-07:
Hallo ManuIch bin im Moment nicht im "Labor".  Test heute Abend wieder.Grüsse Andreas

roema, 2020-11-07:
> Hallo Manu Ich bin im Moment nicht im "Labor".  Test heute Abend wieder. Grüsse Andreas 
> […](#)

Ok!

Vielen Dank!

roema, 2020-11-07:
version0.0.8 should fix this problem

ubx, 2020-11-07:
Hi Manu

With the latest image it works! But I'm a little confused. I thought the legacy protocol is for FLARM only.
If I set Protocol 1 to Legacy then both  OGNTP and FLARM stations are forwarded to the ogn network.

Andreas

roema, 2020-11-07:
Hi Andreas!

C42464 is the id from your ogn basestation and used as callsign.
If you see a packet like 

```
Walpwil-M>APRS,TCPIP*,qAC,C42464:/203741h4703.67NI00713.70E&/A=001541
Walpwil-M>APRS,TCPIP*,qAC,C42464:/205125h4703.67NI00713.70E&/A=001541
Walpwil-M>APRS,TCPIP*,qAC,C42464:>205125h 0.0.8_ESP32 3.93V 
```

 its not an aircraft packet, that are status packets.

Are you sure that your station is sending OGNTP packets? If you've set legacy as protocol-1, that's actually impossible.

I will close the issue as the original problem has been resolved. However, I noticed a second problem in the APRS keepalive package which is fixed with version 0.0.10.

Let me know if your problem still exists !

Manu

closed 2020-11-07 by roema

ubx, 2020-11-08:
Hi Manu

You're right, I completely forgot, the OGN's device protocol was set to legacy.  It works as expected!
Tested with version 0.0.9.

Andreas