roema/Open-Glider-Network-Groundstation #2: Bad sensitivity

roema, 2020-11-09:
The sensitivity of the SX1276 is very limited due to an incorrect configuration.

roema, 2020-12-25:
I found an error in the RX configuration which affects the SX1276. This will be resolved by the end of December. The SX1262 is also being tested.

CazYokoyama, 2021-09-14:
Manuel,
Do you still think SX1276 is less sensitive than RTL-SDR? I installed 2 receiver.

RPI3+RTL-SDR, https://ognrange.glidernet.org/#10R3,max,d30,45.59220_-123.01480,10,#80000040:#008000ff,
t-beam(SX1276), https://ognrange.glidernet.org/#10r4,max,d30,45.60580_-123.03050,10,#80000040:#008000ff,

Both shows the last 30 days and their ranges are double/half difference. Their antennas are both 1.8m long. The antenna of RPI3+RTL-SDR locates the roof of the 2nd floor while another locates the roof of 1st floor.

roema, 2021-09-16:
Hi Caz!

I have tested sensitivity from SX1262 and SX1276 with Flarm (FSK).

SX1276 is better than SX1262.

Unfortunately, both are worse than a good SDR stick.
The two chips were made for Lora and are not particularly good at FSK. 

At Fanet (is Lora), however, they are very good.

We have also found that when cellular base stations are nearby, the sensitivity drops sharply (FLARM).

This can be improved to a certain extent with a good filter and metal box.

Stay healthy!

Kind regards, Manu