roema/Open-Glider-Network-Groundstation #3: FANET messages

Fiddich, 2020-11-16:
Hi,

Our goal is to be also able to warn pg/hg pilots through FANET messages.
Would it be possible to send messages with FANET? While listening primarly on FLARM for receiving position data?
Because almost all FANET devices also send FLARM packages, this seems a logical choice for us. 

In a way that the TTGO only switches to this mode when there is a message for transmit on the queue, which will be limited and controlled by ATC tower.

Thx a lot, the recent OGN developments are very nice.

roema, 2020-11-17:
Hello Fiddich!

Yes, that is possible! It's one of the features I'd like to implement next. Since we only send APRS messages every 5 seconds, there is enough time in between to send FANET packages over the air.

At the moment I don't have a FANET device to test. I have to take a closer look at the Fanet protocol. 

Thank you for your message, I am happy if there is interest in this project.

BPieter, 2020-11-17:
Hi Roema,

A lot of interest. We are currently investigating rolling out a few ground stations. 2
021 goal is livetracking coverage. Messaging would be nice, but its a goal to be complete in 2022. 

There are a few possibilities: 
- using RPi with a LoRa hat, but the development for FANET is ongoing. And this means also 2 antenna's.
- using T-BEAM with your project or;
- GXAirCom project.

As FANET device you can use the GXAirCom project (?) 
it can also act as a groundstation and weatherstation support, but more obvious use is currently as a tracker. 
With the same limitation: not possible to use different protocols at the same time because of the chipset. 
It's either FLARM or FANET.

Would like to help any project, but not a lot of experience yet with C/C++ ... only small modifications.

roema, 2020-11-17:
Hi Pieter!

Thanks! I will take a look at the GXAirCom project.

Today I studied the protocol specification from Fanet and it shouldn't be a big problem to send messages of type 4 (service) and 5 (landmarks). I will implement and test that over the next few days. 

But first I have to organize a FANET receiver in order to be able to test.

Thank you! I am happy about any help and suggestions!

closed 2022-04-26 by roema