roema/Open-Glider-Network-Groundstation #11: Configuration via json file does not work

ubx, 2021-03-08:
Hi Manuel

With the version 0.1.0-17 I'm not able to configure the device via Upload Files with this [config.zip](https://github.com/roema/Open-Glider-Network-Groundstation/files/6099031/config.zip) file. Is has just no effect.
Andreas

CazYokoyama, 2021-03-08:
I have the same problem. Here is the output from /dev/ttyUSB0.

▒S▒▒▒▒k▒▒b`▒@Qu▒9▒8▒▒▒▒3r:▒98h
OGNB-ESP32 FW.REV: 0.1.0-17 DEV.ID: 9cccd34
Copyright (C) 2020 Manuel Roesel. All rights reserved.

Reset reason: 4
SW_CPU_RESET
Free heap size: 213648
Software reset CPU

EEPROM version: 94
SX1262 RFIC is detected.
INFO: TTGO T-Beam rev. 08 is detected.
Flash memory ID: 684016
INFO: GNSS module HW version: 00040007
INFO: GNSS module FW version: 7.03 (45969)
Failed to read file, using default configuration
No WiFi connection information available.
Hostname: OGNB-cccd34
Wait for WiFi connection.
Setting soft-AP configuration ... Ready
Setting soft-AP ... Ready
IP address: 192.168.1.1
UDP server has started at port: 12389

ubx, 2021-03-13:
Tested with version 0.1.0-18, the problem persists.

roema, 2021-03-13:
Hi Andreas!
Yes, i know.

Can you test please

https://jsonformatter.curiousconcept.com/

and then copy the new formatted json data to the config.json file.

It's a bit of a curious bug. I'm looking, unfortunately I don't always have the error.

ubx, 2021-03-13:
Done, the formatter says "Valid (RFC 8259)". Upload Files -> config.json has no effect.

roema, 2021-03-13:
Thanks andreas!

I have found an error! Json buffer to small, if more then one ssid is set.
Can you test this image?

[ognbase.ino.esp32.zip](https://github.com/roema/Open-Glider-Network-Groundstation/files/6134484/ognbase.ino.esp32.zip)

ubx, 2021-03-13:
Tried but it's not working. It seems somehow everything  got messed up.
Should I erase the flash memory and upload from scratch? Do you need trace output?

roema, 2021-03-13:
This image delete the file, before uploading.

Yes, maybe its a good idea to erase the esp32 and upload from scratch.

On Linux 
esptool.py erase_flash

[ognbase.ino.esp32.zip](https://github.com/roema/Open-Glider-Network-Groundstation/files/6134628/ognbase.ino.esp32.zip)

CazYokoyama, 2021-03-13:
The following procedure always works.

./espressif/esptool/esptool.py erase_flash
# install softrf
# connect SoftRF-XXXXXX, open http://192.168.1.1/
# click firmware update and load ognbase.ino.esp32_XXXXXX.bin.

ubx, 2021-03-14:
Hi Manuel

Today I connected the device to my PC and did:

1.erase esp32
2. install softrf
3. install ognbase.ino.esp32_v0.1.0-18.bin
4. connect  to 192.168.1.1
4. upload index.html, update.html and style.css
5. configure config.json with my coordinates, wifi etc
6. upload config.json
7. power cycle esp32
8. eventing looks good
9. make some changes  in config.json (ex. range)
10. upload config.json
11. no effect, i.e. the range did not change
12. reboot (red button)
13. range has changed as wanted!

roema, 2021-03-15:
Hi Andreas!

Thanks for you report and tests!
yes, a config.json upload needs a reboot.

In next version, after uploading config.json the system will reboot automaticly.

closed 2021-03-15 by ubx