roema/Open-Glider-Network-Groundstation #12: Signal Not picking up

bortek, 2021-03-09:
Hi

I had two SoftRF MarkII devices and took one of them and flashed with this Groundstation image file. Configured protocol_1  to OGN and other parametrers. Started it and it looks to be running ok.

Second MarkII device as well as SoftRF Dongle devices are started (also on OGN) and sending their positional data. They see each other. But Groundstation is not picking up any of those two.

What am I doing wrong?

Inside MarkII there are latest T-beam v1.1 modules.

It is connectedto Wifi  

this is how config WebGui page looks like
[image]

roema, 2021-03-09:
Hi Bortek!
Thanks for your Message!

Give me some time, i will test it tomorrow!
Manu

bortek, 2021-03-10:
I am glad you answered. Looking forward for your reply and let me know if you need any additional info or help with testing.

roema, 2021-03-12:
Hi Bortek!
I can confirm, ogn protocol is not working.

I hope, i can release this weekend  a new version with the fix.
Thanks for waiting!

Manu

bortek, 2021-03-12:
thanks for testing it! Keep up a good work!

roema, 2021-03-12:
Thank you Bortek!

I have found the bug! -> RX Setup befor loading config file

Please test version 0.1.0-18 and give me short feedback please.

Thank you for testing. I will upload new image in few minutes.

Manu

bortek, 2021-03-13:
One more finding with the v-17. Only Legacy/Flarm is working.

FANET is not working either.

Will test v-18 now.

bortek, 2021-03-13:
Good find!

Now testing with Version 0.1.0-18

OGN works!!!

FANET also works.

Good work!!!

closed 2021-03-13 by bortek