roema/Open-Glider-Network-Groundstation #14: Does not connect to WiFi AP

CazYokoyama, 2021-03-15:
I can connect ognbase when it acts as an AP, i.e. 192.168.1.1, but it does not connect AP. I configure SSID and password which used to work. As a result, it does not feed information to ogn server.
Does anyone have the same symptom?

roema, 2021-03-22:
Hi Caz!
Sorry for the late answer, at the moment i am a little bit stressed.

Its correct, your stations connects to your ap but doesnt send aprs messages to ogn network?
Can you check debug messages over udp?

Are all necessary fields set? (without gps you need lat, lon, alt, geoid)

CazYokoyama, 2021-03-22:
> Hi Caz!
> Sorry for the late answer, at the moment i am a little bit stressed.
> 
> Its correct, your stations connects to your ap but doesnt send aprs messages to ogn network?
No. My stations does not connect to my ap. SSID and its password are same as before.

> Can you check debug messages over udp?
Will do.

> 
> Are all necessary fields set? (without gps you need lat, lon, alt, geoid)
I believe so. Red LED of my t-beam(i.e. GPS fix signal) is flashing.

-caz

CazYokoyama, 2021-03-23:
> > Can you check debug messages over udp?
> > Will do.
How can I do it? udp_server.py fails as

[caz@diana2 ogn-receiver]$ python3 udp_server.py 12000
Traceback (most recent call last):
  File "udp_server.py", line 19, in <module>
    serverSock.bind((UDP_IP_ADDRESS, UDP_PORT_NO))
OSError: [Errno 99] Cannot assign requested address

when UDP_IP_ADDRESS = "192.168.1.200". As I said, ognbase act as an AP. Its IP is 192.168.1.1. Even UDP_IP_ADDRESS = "192.168.1.1", I have the save error message.

JacquesGr, 2021-03-27:
Hi Caz,
You need to set the machine that has the UDP receiver to x.x.x.200 in my case my subnet is 192.168.86.0/24 so I setup pc to 192.168.86.200.

Then in the python script you put your IP address eg: UDP_IP_ADDRESS = "192.168.86.200"

Then you run: python3 udp_server.py 12000

roema, 2021-03-28:
Hello Caz! 

Please excuse the late news, I'm a little stressed at the moment.

JacquesGr is right about the configuration. 

A serial connection would be best to see what is going wrong with the WiFi. The beam checks the connection to the access point every 10 minutes; if it is in AP mode, it restarts and tries again to connect to the access point.

Have you tried deleting the beam with the flash tool and reprogramming it from scratch?

CazYokoyama, 2021-04-17:
Found a root cause.
- Don't configure more than one AP

closed 2021-04-17 by CazYokoyama