roema/Open-Glider-Network-Groundstation #16: Zabbix config

JacquesGr, 2021-03-29:
For the Zabbix monitoring

I assume we create a file called zabbix.txt with:
xxx.xxx.xxx.xxx (IP of zabbix server)
10051
ogn_base

Where do we save the file?
I tried to upload it via the OGNBS gui but I dont see any traffic hitting my zabbix server

Also in the case above, is ogn_base the key?

Thanks

roema, 2021-03-29:
Hi Jaq!
In new version, use config.json. There is a section with zabbix.
I have forgotten to update the readme, now its correct, please excuse me!

At the moment, its very rudimentary. 
The beam is sending zabbix traps **without encryption**!

ogn_base is the name of the device in zabbix server. Configure the device.type as zabbix trapper and create items like battery voltage.

I've attached a few screenshots for you. 
If you have any questions, please let me know at any time!!!

[image]
[image]
[image]

roema, 2021-04-16:
Sorry for the delay. I will test Zabbix feature this weekend!

roema, 2021-10-31:
Use config.json to configure zabbix values.
Make sure your server is pingable.

commit 4fdd0308 (2021-10-31): zabbix trap sender fixed #16
closed 2021-10-31 by roema