roema/Open-Glider-Network-Groundstation #17: Status and version info heartbeat missing

balazs-h, 2021-04-19:
Feature request: 
Please send version info to aprs server. 

Correct me if Im wrong, but I think that the software does not send version information and status to the aprs server. Thats why if you look at the groundstation map, at the TTGO groundstations you dont see version info. 

Example: 
https://ognrange.glidernet.org/#DIOSD,max,all,47.40436_18.94685,17,#80000040:#008000ff,

In the aprs log this is what I see as TTGO heartbeat: 

DIOSD>APRS,TCPIP*,qAC,C7CDF0:/053854h4724.26NI01856.59E&/A=000393

This is the Raspberry heartbeat followed by a status with cpu, memory and temperature info:

DIOSD-RPI>APRS,TCPIP*,qAC,GLIDERN5:/053827h4724.28NI01856.74E&/A=000361
DIOSD-RPI>APRS,TCPIP*,qAC,GLIDERN5:>053827h v0.2.6.RPI-GPU CPU:0.5 RAM:238.0/454.6MB NTP:0.8ms/-9.2ppm +47.1C 0/0Acfts[1h] RF:+72+88.8ppm/-4.69dB

Anyways, It seems that the system works like this, but I think this would be important to send this status line aswel to the aprs server. So beside zabbix server we could monitor the groundstation through the aprs server also. And the most important: The groundstation map would look NICE!!! :)

What do you think?

roema, 2021-04-19:
Hi Balazs!

Thanks for you message!
When i look glider-radar.com to your station, it shows messages like

```
 19.04.2021 20:51:38+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:>185136h 0.1.0-20_ESP32 0.00V
19.04.2021 20:53:13+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/185311h4724.26NI01856.59E&/A=000393
19.04.2021 20:58:15+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/185812h4724.26NI01856.59E&/A=000393
19.04.2021 21:03:15+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/190313h4724.26NI01856.59E&/A=000393
19.04.2021 21:08:16+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/190814h4724.26NI01856.59E&/A=000393
19.04.2021 21:13:17+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/191315h4724.26NI01856.59E&/A=000393
19.04.2021 21:18:18+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/191816h4724.26NI01856.59E&/A=000393
19.04.2021 21:21:38+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:>192136h 0.1.0-20_ESP32 0.00V
19.04.2021 21:23:19+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/192317h4724.26NI01856.59E&/A=000393
19.04.2021 21:28:21+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/192818h4724.26NI01856.59E&/A=000393
19.04.2021 21:33:21+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/193319h4724.26NI01856.59E&/A=000393
19.04.2021 21:38:23+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/193820h4724.26NI01856.59E&/A=000393
19.04.2021 21:43:23+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/194321h4724.26NI01856.59E&/A=000393
19.04.2021 21:48:24+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/194822h4724.26NI01856.59E&/A=000393
19.04.2021 21:51:38+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:>195136h 0.1.0-20_ESP32 0.00V 
```
Ogn Station will send status messages every 30 minutes. Its no problem to send the status messages after every keepalive packet.

Unfortunately, the esp32 does not provide as much status data as the CPU and RAM load. The status message is therefore kept very brief.

balazs-h, 2021-04-20:
Hi Manu,
I Think maybe a letter v and h is missing from the beginning of the
software version. Thats why it shows no version info? Can you make a try?

Like: hLilygo-TTGO-TBeam v0.1.0-20_ESP32

Description:

http://wiki.glidernet.org/wiki:ogn-flavoured-aprs

Receiver status beacon:

OGN3FC859>OGNTRK,qAS,LZHL:>093215h h00 v00 9sat/1 164m 1002.6hPa
+20.2degC 0% 3.34V 14/-110.5dBm 1/min

Description:
h00 = hardware version (not being filled yet)
v00 = software version (not being filled yet)
9sat/1 = GPS status: 9 satellites, FixQuality=1 (GPS)
164m = altitude by GPS
1002.6hPa = pressure
+20.2degC = temperature by baro, if not present then by CPU (but not very
accurate then)
0% = humidity, no sensor yet…
3.34V = battery voltage, here battery divider not installed thus simply the
3.3V VCC, subject to innacuracies due to Vref of the CPU.
14/-110.5dBm = receiver noise level
1/min = number of packets received per minute (gives an idea of (radio)
traffic intensity)
External Links

balazs-h, 2021-04-20:
Hallo Manuel!!!
SUCCESS!!!

Here is the correct status/version format: v0.1.0.20-TTGO
2 comments:

1. first char has to be v
2. There can be 3 dots . and one slash -  in the version string

I tried with multiple slashes. Like v0.1.0.20-TTGO-TBeam
I didnt work.

3. Also I think the Maximal characters can be 13 characters.
So this is the ideal version string:
v0.1.0.20-TTGO

Heres the correct heartbeat and status line, that worked for me:

DIOSD>APRS,TCPIP*,qAC,C7CDF0:/115609h4724.26NI01856.59E&/A=000393

DIOSD>APRS,TCPIP*,qAC,C7CDF0:>115609h v0.1.0.20-TTGO 0.00V

Hellinger, Balazs ***@***.***> ezt írta (időpont: 2021. ápr. 20.,
K, 6:34):

> Hi Manu,
> I Think maybe a letter v and h is missing from the beginning of the
> software version. Thats why it shows no version info? Can you make a try?
>
> Like: hLilygo-TTGO-TBeam v0.1.0-20_ESP32
>
>
> Description:
>
> http://wiki.glidernet.org/wiki:ogn-flavoured-aprs
>
>
> Receiver status beacon:
>
> OGN3FC859>OGNTRK,qAS,LZHL:>093215h h00 v00 9sat/1 164m 1002.6hPa +20.2degC 0% 3.34V 14/-110.5dBm 1/min
>
> Description:
> h00 = hardware version (not being filled yet)
> v00 = software version (not being filled yet)
> 9sat/1 = GPS status: 9 satellites, FixQuality=1 (GPS)
> 164m = altitude by GPS
> 1002.6hPa = pressure
> +20.2degC = temperature by baro, if not present then by CPU (but not very
> accurate then)
> 0% = humidity, no sensor yet…
> 3.34V = battery voltage, here battery divider not installed thus simply
> the 3.3V VCC, subject to innacuracies due to Vref of the CPU.
> 14/-110.5dBm = receiver noise level
> 1/min = number of packets received per minute (gives an idea of (radio)
> traffic intensity)
> External Links
>
>
>
>
>> Hi Balazs!
>>
>> Thanks for you message!
>> When i look glider-radar.com to your station, it shows messages like
>>
>>  19.04.2021 20:51:38+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:>185136h 0.1.0-20_ESP32 0.00V
>> 19.04.2021 20:53:13+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/185311h4724.26NI01856.59E&/A=000393
>> 19.04.2021 20:58:15+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/185812h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:03:15+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/190313h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:08:16+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/190814h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:13:17+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/191315h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:18:18+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/191816h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:21:38+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:>192136h 0.1.0-20_ESP32 0.00V
>> 19.04.2021 21:23:19+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/192317h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:28:21+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/192818h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:33:21+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/193319h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:38:23+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/193820h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:43:23+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/194321h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:48:24+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:/194822h4724.26NI01856.59E&/A=000393
>> 19.04.2021 21:51:38+02:00: DIOSD>APRS,TCPIP*,qAC,C7CDF0:>195136h 0.1.0-20_ESP32 0.00V
>>
>> Ogn Station will send status messages every 30 minutes. Its no problem to
>> send the status messages after every keepalive packet.
>>
>> Unfortunately, the esp32 does not provide as much status data as the CPU
>> and RAM load. The status message is therefore kept very brief.

roema, 2021-04-20:
Hey Balazs!

Thanks for your tests!
I will fix this in next release!

Best regards, Manu

roema, 2021-08-01:
Hi Balazs!

Sorry for the delay!
Please check latest version and close issue if fixes

roema, 2022-04-25:
Hey @balazs-h !
I hope you are well.
I will close this issue now, if you have problems, please open a new one.

Thanks, regards, Manuel

closed 2022-04-25 by roema