roema/Open-Glider-Network-Groundstation #21: saving is not possible

hem0773, 2021-07-14:
if you click on the "save button", the data is lost and the "old" start page appears where you only had to download the files

My board is V1.1
[image]

roema, 2021-07-17:
Hi @hem0773 !

Thanks for you issue!
The last few weeks i was a bit stressed, so i couldn't do anything for a longer period of time. From month on I have more time for the project. I will look into the problem. 

Thanks for your patience!

Manu

hem0773, 2021-07-20:
OK thanks
But no data is received

What am I doing wrong

[image]

roema, 2021-07-31:
should be fixed in latest version 0.1.0-21

Thanks to Nick.

closed 2021-07-31 by roema