roema/Open-Glider-Network-Groundstation #23: status packet has battery voltage

CazYokoyama, 2021-08-29:
My OGN receiver, 10r4, 
https://www.gliderradar.com/center/45.60719,-123.06490/zoom/12/time/15/maptype/satellite is t-beam and powered by solar panel. I would like to monitor weather it is in good shape and expect keep running until next morning when the solar panel produces power.

roema, 2021-09-11:
Hey Caz!

I dont understand, what do you mean with "status packet has battery voltage"

CazYokoyama, 2021-09-11:
Manuel,
Disregard and will close. Glider RADAR - 10r4 Details of https://www.gliderradar.com/details/main/sid/1048458 has

Latest Status:	110548h v0.1.0-22-ESP32 3.41V
Receive Time:	09/11/2021 4:05:50 AM-07:00

10r4 is
- Lilygo t-beam
- 1.8M long vertical antenna
- 3000mAh LiPo
- 2W solar panel

It stops sending heartbeat from time to time. I wanted to know whether it is in sleep mode or totally die.

closed 2021-09-11 by CazYokoyama