roema/Open-Glider-Network-Groundstation #26: Can't build ognbase.ino.esp32_TTGO_V0.1.0-26.bin from source

CazYokoyama, 2021-11-09:
TTGO PaxCounter has arrived. The binary can't be installed by web IF because Its pre-installed software, PaxCounter does not have web IF. Source build by arduino has error as the png file. I am glad if someone tells me what I missed.

[image]
[image]
[image]

roema, 2021-11-10:
Hi Caz!Usw the same compile options and board like T-Beam. The only different ist the psram. Set it to disable.

CazYokoyama, 2021-11-10:
With the direction above, I successfully build, install, configure PaxCounter.

closed 2021-11-10 by CazYokoyama

schenlap, 2022-03-26:
Ich habe hier das selbe Problem.

```
Sketch uses 1338226 bytes (102%) of program storage space. Maximum is 1310720 bytes.text section exceeds available space in board

Global variables use 60696 bytes (18%) of dynamic memory, leaving 266984 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board T-Beam.
```

With enabled psram i get 108% instead of 102%.

[image]