roema/Open-Glider-Network-Groundstation #27: Disconnect after some time later such as 20 min.

CazYokoyama, 2021-11-10:
Here is the its output from USB.

OGNB-ESP32 FW.REV: 0.1.0-26 DEV.ID: 25b1ebac
Copyright (C) 2020 Manuel Roesel. All rights reserved.
Reset reason: 0
POWERON_RESET
Free heap size: 279356
Vbat power on reset
WARNING! User defined settings are not initialized yet. Loading defaults...
WiFi config changed.
.6.6.6.6.6.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1
Can not connect to WiFi station
.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1.1
Can not connect to WiFi station
.1IP address: 192.168.0.126
Hostname: OGNB-b1ebac
UDP server has started at port: 12389
SX1276 RFIC is detected.
INFO: TTGO T3 rev. 0 is detected.
Flash memory ID: C84016
Starting NTP UDP
Local port: 2390
#1 NTP server's IP address: 95.216.147.242
sending NTP packet...
Reply packet received, length=48
Seconds since Jan 1 1900 = 3845554380
Unix time = 1636565580
The UTC time is 17:33:00
Failed to open key file, using default key!.
IV file wrong size, using default key!.
using AES key
2B7E151628AED2A6ABF715889CF4F3C
using AES iv
0000000000000000OGN connection failed
OGN connection failed
OGN connection failed
OGN connection failed
....

It shows the screen in attachment for a while and disconnect.
[image]

Do I configure something wrong?

roema, 2021-11-11:
Hi @CazYokoyama !

I have observed this last weeks. I will check and report!
Thanks for your tests!

CazYokoyama, 2021-11-12:
Not disconnected forever. It repeats disconnect and connect.

closed 2022-01-09 by CazYokoyama