roema/Open-Glider-Network-Groundstation #28: multiple APs are defined

CazYokoyama, 2022-01-02:
5 APs are defined by the following code.
String ogn_ssid[5];
String ogn_wpass[5];

@roema Do you expect only one AP and don't want to accept multiple APs? If so, you may define them as not an array. If you want to accept multiple APs. I'll make a patch.

roema, 2022-01-02:
Hello @caz! Multiple APs are supported. You have to enter it in config.json. Only the first SSID is used in the web interface.
```
{
   "wifi":{
      "ssid":[
         "ssid",
         "xxxxxxx",
         "xxxxxxx",
         "xxxxxxx",
         "xxxxxxx"
      ],
      "pass":[
         "password",
         "xxxxxxx",
         "xxxxxxx",
         "xxxxxxx",
         "xxxxxxx"
      ],
      "timeout":12000
   },
```
On boot, the esp will test the ssid's. if no connection to one of the five is possible, he will create acces point...

CazYokoyama, 2022-01-02:
@roema Correction.
If you want to accept multiple APs. I'll make a patch.
----
If you want to accept multiple APs. I'll make a patch which simplify the code by WiFiMulti().

roema, 2022-01-02:
Hey @CazYokoyama !

Yes, if you have time!

CazYokoyama, 2022-01-03:
https://github.com/roema/Open-Glider-Network-Groundstation/pull/29

roema, 2022-01-18:
Hi @CazYokoyama !

Thanks a lot! I will test this weekend!
 Kind regards, manu

roema, 2022-01-31:
#29 Thanks Caz

closed 2022-01-31 by roema