roema/Open-Glider-Network-Groundstation #32: oled wont work with heltec lora 32 v2

qubolino, 2022-02-24:
just got a board from heltec but oled isn't showing anything. any idea what to change in the source?

roema, 2022-02-24:
Hi!

Heltec boards are not tested yet. I have no board from heltec, so i cannot give you informations about, sorry.

qubolino, 2022-02-24:
aren’t the specs enough?

qubolino, 2022-02-25:
what i can say is that  i have the disply working with https://github.com/pjalocha/esp32-ogn-tracker and flags

```#define PIN_I2C_SCL GPIO_NUM_15   // SCL pin
#define PIN_I2C_SDA GPIO_NUM_4    // SDA pin
#define OLED_I2C_ADDR 0x3C        // I2C address of the OLED display
#define PIN_OLED_RST GPIO_NUM_16  // OLED RESET: low-active
```

maybe there's a simple way of building ognbase according to these pins/address

roema, 2022-02-26:
You can test it, maybe it will be work. You should also check the pins from lora chip and gps if available.

qubolino, 2022-03-01:
basestation essentially works out of the box besides display on my side

closed 2022-04-26 by roema