roema/Open-Glider-Network-Groundstation #34: No data from GPS even when fix led blinking

dziedzic99, 2022-03-26:
Note that some of the TTGO T-Beam devices come with GPS module preprogrammed in a way that will not work with this software.
**What fixed it for me:** set board in Serial Passthrough mode between Serial and Serial1, both at 9600 baudrate. Then use `ubxtool -p RESET -f /dev/YOURSERIALPORT` to reset GPS to default settings. Afterwards flash this software and it will work.

Suggested fix in the software: add GPS module reset command at boot.

schenlap, 2022-04-24:
Same problem here. After `ubxtool -p RESET -f /dev/YOURSERIALPORT` the red led ist not flashing any more.
`GNSS_version()` returns `GNSS_MODULE_NONE` on TTGO T-Beam T22 V1.0 (GPS TX=IO34, RX=IO12).

dziedzic99, 2022-04-24:
Try resetting it again? It took a few tries for mine to catch up.

roema, 2022-04-25:
Sorry for the late answer, i am very busy at the moment.
Thanks for information, i will try to fix for next version!

I think next weekend i am ready to publish new version.

closed 2022-04-26 by roema